VoronDesign/Voron-Documentation #26: AC Wiring AWG to mm2 conversion

jhreg123, 2021-03-02:
Hi,

Thanks for putting these docs together, they're amazing. One small thing I spotted is that the conversion from AWG to mm2 for mains AC wiring in the tips section of Electrical Wiring seems off compared to most conversion tables I've seen.

https://www.cse-distributors.co.uk/cable/technical-tables-useful-info/awgmetric-conversion/
https://www.multicable.com/resources/reference-data/cross-reference-awg-to-mm2/

I'm not 100% sure which value is meant to be the correct one otherwise I'd update it myself.

Thanks!

javicarrera, 2021-08-12:
Just for your information I leave you the standard cable gauges in mm2.

In black metric, in blue AWG
https://upload.wikimedia.org/wikipedia/commons/3/32/Comparison_wire_gauge_sizes.svg

For our application these are the relevant ones:
0.14 mm2 > AWG26
0.25 mm2 > AWG24

0.5 mm2  < AWG20
0.75 mm2  < AWG18
1 mm2 < AWG17
1.5 mm2  < AWG15
2.5 mm2 < AWG13
4 mm2  < AWG11
6 mm2 < AWG9

Regulated or standardized sections start with 0.5mm2. Smaller sections can slightly vary.

bdbell, 2021-08-29:
Updated all references found in PR #84.  If any more are found that are wrong please provide specific reference.

closed 2021-08-29 by bdbell