VoronDesign/Voron-Documentation #31: Troubleshooting checklists

ehellman, 2021-03-15:
I think there should be a "common issues" section with troubleshooting checklists for each printer/component in the docs. I just encountered something that has to be a common issue with extrusion:

I opened the door to my AB after hearing some clicking noises, expected a clog but there was no clog at all. used a small screwdriver to poke at the nylon gear and I could move it back and forth a little bit, affecting the bondtech gear too. So I took everything out and placed a 1mm thick M5 washer at the foot of the nylon gear, put everything back - viola! No more clicking and extrusion is 👌 This could be under Afterburner > Clicking/knocking during extrusion, grouped together with clog/jam/skipping steps as a possible cause of the clicking.

However, me being new to the Voron universe I don't really feel I could make a PR for this already since there are many people with much more experience with issues that can arise that could get this started.

What do you guys think?

Haszard, 2021-12-28:
I am experiencing filament skipping as well. Just finished my Formbot Voron 2.4 Kit  about a month ago. Ordered all the printed parts from a pro printing service, really good quality, as of what i could tell. 

Just like ehellman i have been troubleshooting from day one because of this. The one thing that i found, really to make a difference, was switching out the V6 Hotend that came with the kit, to a Phaetus Dragon HF hotend. 
I am convinced that the V6 in the kit was a knockoff, and suffered from severe heat creep also. 

But... The skipping persists to this day, although it is much much less now. Partial victory at the least. :-) 

I now suspect the clockwork extruder to have a defect of some kind, weather it to be a design fault, or maybe me putting something together wrong, although i have taken it apart once to inspect. But i will da so again, in case i missed something. 

A Troubleshooting section would be much appreciated. 

Will update this post with my findings.

sundansx, 2021-12-28:
I did not fix my afterburner skipping problem.  I have since then tried everything that others have suggested on Discord, upgraded to a phateus dragon standard flow, and did all the things in [AndrewEllis93's excellent guide](https://github.com/AndrewEllis93/Print-Tuning-Guide#first-layer-squish) (thanks).  I have moved on to a LDO kit galelio oribital extruder, as a lot of people say those have more mechanial advantage.  So far, that seems to be doing much better.  I do really like the afterburner design, tho.

Haszard, 2021-12-28:
My backup plan is to try an Bondtech LGX if everything else bears no fruit. Any thoughts on that? 
I have no personal experience with this extruder, i am coming from a mildly modified Ender 3 pro ( Microwsiss all metal hotend and skr mini e3 ect. ) no comparison whatsoever to this projekt. :-)

sundansx, 2021-12-28:
I have never tried a bondtech, although I hear they are good, if not a bit expensive.  Another option is the The LDO Orbiter V1.5 Dual Drive Direct Extruder.  That is another that is about half of the bondtech price and is a full solution with no printer parts like the bondtech - it is also said to be pretty light, like the galileo.

Haszard, 2021-12-28:
Great info, thank you! I newer read up on that one, it will be taken into consideration it it comes that far.   
In my search tonight i found this: 
https://github.com/AndrewEllis93/Print-Tuning-Guide#extruder-skipping
It has a very comprehensive list of possabilites for filament skip. Guess i will have to go through it one by one. 
Maybe your solution is in there somewhere too.

molekm, 2022-05-18:
Same issue here, I've ditched 3d printed clockword to LDO Orbiter. No point wasting time as it worked fine for 900 hours and then it started. 3d printed parts for such speeds and such mechanical part is not a good idea when you have ready to buy solutions that are MUUUUUUUCH better. Ditch clockwork and switch to Orbiter. You will thank me later.