VoronDesign/Voron-Documentation #47: Confusing wording in mini12864 instructions

arkeet, 2021-05-05:
https://docs.vorondesign.com/build/electrical/mini12864_klipper_guide.html

A user recently had a problem getting the display working, misunderstanding the header flip image and thinking "yes" meant to flip one connector, and "no" meant to not flip the other. Suggestion:
* Change the words the image to say "correct" and "wrong" or something similar.
* Change step 2 to say something like "Rotate _both_ connector housings 180 degrees and insert back onto pins."

gihad, 2021-05-06:
I'd suggest also changing the first sentence from `NOTE: The connector housing only` to `NOTE: The 2 connector housings only`

bdbell, 2021-08-29:
Updated wording in PR #84.

closed 2021-08-29 by bdbell