VoronDesign/Voron-Documentation #63: Update Documentation to use ssh via CMD/Terminal and FileZilla instead of the various software mentioned

ChaosBlades, 2021-06-27:
`ssh pi@octopi.local` is supported by Windows via Command Prompt and by MacOS/Linux via Terminal. No reason to list a bunch of different software on different platforms when the same thing works on all platforms.

You can transfer files via a GUI program on Windows, Linux, and MacOS with FileZilla. Use that instead of mentioning different software for each platform.

These two thing will help to clean up documentation a little bit by keeping things the same between platforms.

shiftingtech, 2021-08-29:
Unfortunately, I can't fully agree.
While I agree with your point about simplification, I've also seen a lot of the...less experienced... users get pretty tangled up trying to use the command line ssh and scp.

ChaosBlades, 2021-08-29:
I am not sure what you are getting at. It is EXACTLY the same. Except you don't need to download or install any programs. MacOS, Linux, and BSD users are already going to be using there own versions of cmd which also work the same exact way in this case. Open cmd/terminal and type `ssh pi@octopi.local` and it is the exact same experience as putty, except native.

FileZilla is a GUI program for scp so no command line.

ChaosBlades, 2021-08-29:
It is literally just adding more steps by installing putty. Using cmd instead is literally just removing unnecessary steps allowing less opportunity for less experienced users to have issues.

shiftingtech, 2021-08-29:
well, to give you one simple example,
1)ssh into a pi.  do stuff
2)decide to reflash the sdcard
3)try to ssh in again.

in putty, you get a nice little dialog:  the host identification has
changed, are you sure you want to connect?
in command line ssh (at least with the default settings used on windows),
you get a hard stop, and need to know some other, completely esoteric
commands to erase the old key.

ChaosBlades, 2021-08-29:
Hardly esoteric... and the same experiences you will get on MacOS, Linux, and BSD creating consistency between platforms which is the point of this issue.

ssh-keygen -R "you server hostname or ip"

bdbell, 2021-08-30:
I would be willing to add documentation on how to use the native SSH capabilities of Windows.  However I'm not going to hard cut and get rid of the instructions for PuTTY either.  If you are using the native SSH capabilities of Windows you are obviously familiar with it, but we run into LOTS of people who are not so there has to be some flexibility.

ChaosBlades, 2021-08-30:
If that is the case then we need none native solutions for OSX and Linux as well. Unless what you are saying is that windows users are dumber than osx and Linux users.

I would say Remmina would be the choice for Linux. I don't use apple and everyone I know uses terminal for ssh so I have no input there.