VoronDesign/Voron-Documentation #67: Resourcing Guide using s.click.alieexpress.com referrer spam

ChaosBlades, 2021-06-29:
How to block it in Google Analytics:
https://botcrawl.com/block-s-click-alieexpress-com-referrer-spam/

What it is:
https://en.wikipedia.org/wiki/Referrer_spam

bdbell, 2021-06-29:
Can you please provide an example?

ChaosBlades, 2021-06-29:
https://s.click.aliexpress.com/e/_dWCqSj2

ChaosBlades, 2021-06-29:
This is the Google Analytics link on the sourcing guide for the V6. The one I linked above
https://www.google.com/url?q=https://s.click.aliexpress.com/e/_dWCqSj2&sa=D&source=editors&ust=1624995118453000&usg=AOvVaw21Aes6qYrwWhxZXTj9a4t7

There are a couple like this Dragon Hotend that are the correct link to Ali
https://www.google.com/url?q=https://www.aliexpress.com/item/1005002477263672.html?&sa=D&source=editors&ust=1624995118453000&usg=AOvVaw0gCRvuKcsbD3LdXRM1yntm

raymondh2, 2021-06-29:
s.aliexpress.com denotes affiliate links IIRC. Removing affiliate links isn't really something we want to do completely. But most guides should list both.

ChaosBlades, 2021-06-29:
I looked it up, it is an affiliate link. I guess ali is doing double duty with these links? 

Seen it was in my DNS blocklist and looked it up. I'll keep it in the block list and just turn my VPN on to bypass when using the Sourcing Guide.

closed 2021-08-29 by bdbell