VoronDesign/Voron-Documentation #130: PL-08N2 easily modified for compatibility with new tool head carriage.  2.4R2 - Trident...going forward.

PhilBaz, 2021-12-30:
Thought this might be useful for all the people with PL-08N2 that they want to keep. My works excellent. Also, since it is the cheapest probe option by far.

In going to the MGN12 i realized that my PL-08N2 is to tall, and now understand it is no longer supported from trident forward. 

But!!! The top of the PL-08N2  housing can easily be removed with pliers to be cut down so it fits with the new MGN12 trident carriage. I think this should be part of the documentation or at least common knowledge. Why waste all those probes if they are working well, and then shell out more cash for Omorons.
[image]

bdbell, 2022-03-23:
This seems clear enough.  Could you submit a PR to add that to the community documentation?

PhilBaz, 2022-04-09:
@bdbell Ill figure out how to do that in the next week or so. Im new to git and the whole PR thing.