VoronDesign/Voron-Documentation #184: Uncomment in [gcode_macro G32] missing from Software Configuration docs

jullit31, 2022-03-15:
The [Updating Printer Specific Settings](https://docs.vorondesign.com/build/software/configuration.html#updating-printer-specific-settings) section doesn't tell you to uncomment in the `[gcode_macro G32]` section ( which **is** mentioned in the default `printer.cfg` as "things to check"). This is very bad, as failing to do so will result in the nozzle crashing into the z-endstop when starting the first print. (Don't ask how I know 🙈)

shiftingtech, 2022-03-15:
doesn't tell you to uncomment *what* ?  what printer are we even talking about?

jullit31, 2022-03-15:
Oops, Voron 2.4.
There is only one thing to uncomment in the `[gcode_macro G32]`. Specifically I am referring to [this](https://github.com/VoronDesign/Voron-2/blob/9d0ef05a5bb4ff18f28bf5bf0fc73765c0870bd7/firmware/klipper_configurations/Octopus/Voron2_Octopus_Config.cfg#L528) and the following lines in the `printer.cfg` (here the Octopus config as an example, but of course it's the same for other boards).