VoronDesign/Voron-Documentation #329: Missing explanation for component removal on mini12864 displays

manisto, 2023-01-17:
On the [Mini12864 documenation](https://docs.vorondesign.com/build/electrical/mini12864_klipper_guide.html), it tells you to remove certain components from the display board. In the case of BTT, it's R1 and C6. However, the display seems to work fine (at least for BTT), so an explanation of why this is needed would be nice. Also, if this is important, the assembly manual would benefit from a link to that documentation page in the relevant assembly steps.

shiftingtech, 2023-01-17:
Point of clarification:  it tells you that if you are using the btt display on a fysetc spider, you need to remove those components.  Are you running that specific combination?

manisto, 2023-02-03:
Hello, sorry for the late reply. I thought I read the documentation thoroughly, but it seems like I still missed a few key points. Both my display and my board is BTT, so I do not have to do anything to my display. So the only thing I guess would be nice would be an explanation on why you need to remove components from your display if you have a Fysetc Spider board.