VoronDesign/Voron-Documentation #332: Link to sensorless homing guide is broken on docs.vorondesign.com

EDGe-17, 2023-02-07:
The link is broken to sensorless homing guide in the Voron documentation website. 
Link found on https://docs.vorondesign.com/build/startup/

-Incorrect path https://docs.vorondesign.com/build/startup/Voron-Documentation/blob/main/community/howto/clee/sensorless_xy_homing.md 

-Correct path https://docs.vorondesign.com/build/startup/Voron-Documentation/blob/main/community/howto/clee/sensorless_xy_homing.md 

[image]

timmit99, 2023-02-07:
@EDGe-17 both of the links above dont work.

timmit99, 2023-02-07:
I think you meant to put:

- Correct part https://docs.vorondesign.com/community/howto/clee/sensorless_xy_homing.html

nemgrea, 2023-02-09:
https://github.com/VoronDesign/Voron-Documentation/pull/333

closed 2023-02-09 by nemgrea