VoronDesign/Voron-Documentation #340: Initial startup Switchwire Motor Configuration Z direction is inverted

sree314, 2023-03-02:
At this [link](https://docs.vorondesign.com/build/startup/#sw), the first column in this [table](https://docs.vorondesign.com/build/startup/images/SW-motor-configuration-guide.png) has the Z arrows in the opposite direction.

If all is ok, then on the Switchwire, X should go right and Z should go _down_ (instead of up, as the current picture shows).

nemgrea, 2023-03-02:
the image is correct, however the explanation of what the image is showing needs to be made more clear, the image shows positive MOVEMENT directions NOT necessarily the homing direction. obviously this being included in the homing check section does not make that clear, we will work on figuring out how to communicate this better and prevent this confusion from tripping up others, thanks for the report!

sree314, 2023-03-02:
Ah, okay. Indeed, under that interpretation, the image is correct. Thanks!