VoronDesign/Voron-Documentation #355: Relative reference calculation in secondary printer tuning is incorrect

Michael-Jacobsen, 2023-05-02:
In the secondary printer duning doc where it referes to Mesh Points and Relative Reference Index there is formula for calculating the reference index so it is the point at the center of the bed. It currently suggests using  ((x points * y points) - 1) / 2 
On a 3x3 mesh that gives an index of 4 which is on the edge of the bed. With larger mesh numbers the error is lower but it is still the probe point next to the center of the bed rather than the center. 
The formula sould be  ((x points * y points) + 1) / 2 
On a 3x3 mesh that gives us probe point 5 or the center of the bed mesh.
On a 5x5 mesh that gives us probe point 13 (the 3rd point on the 3rd row) or the center of the mesh. point 1 to 5 is row one, point 6 to 10 is row two, point 11 to 15 is row three and the middle point is 13.

shiftingtech, 2023-05-02:
The points are indexed from zero.  For example  for the 3x3 mesh, the points are

6   7   8
3   4   5
0   1   2

closed 2023-05-02 by shiftingtech

Michael-Jacobsen, 2023-05-03:
OK, I had thought of that. It was just that in the klipper docs for bed mesh and under relative reference index where they talk about selecting the point at the center of the bed they have an example of;
[bed_mesh]
speed: 120
horizontal_move_z: 5
mesh_min: 35, 6
mesh_max: 240, 198
probe_count: 5, 3
relative_reference_index: 7

which again would put the index on the edge of the bed if its indexed from zero.

shiftingtech, 2023-05-03:
No, it would not.

Sorry for the bad formatting,  but...

10  11 12  13  14
5  6  7  8  9
0  1  2  3  4

Michael-Jacobsen, 2023-05-03:
Right you are. The 5x3 mesh was throwing me off.
Thankyou.