VoronDesign/Voron-Documentation #360: Easier Klipper flash procedure for BTT Octopus.

cmdremily, 2023-05-11:
The documentation notes "don't power the Pi from the Octopus as it's more tedious (due to power cycles)" and that there are two power cycles of the BTT board.  It's not true that it's more tedious, and the power cycles aren't necessary. Simply hold down the reset button next to the type-c connector on the edge while inserting or removing the `BOOT0` jumper.  No power cycles are necessary.

By holding down the reset button, the MCU is kept in reset and it's not executing any code. All of the GPIO pins are in HiZ, i.e. nothing is driven. Plugging the jumper is safe in this state (as long as you don't roll a 1 on the dexterity check and plug the jumper into a random other pin pair, but a power off won't save you from that either). Upon releasing reset, the MCU executes the reset vector, with BOOT0 bridged it goes into dfu bootloader. Same thing for exiting dfu bootloader, just hold reset button and unplug jumper.

shiftingtech, 2023-05-11:
I strongly disagree with this proposal.  People trying to insert the jumper hot are going to short random things and wreck their boards.

If you're comfortable doing it that way (and have the steady hands for it, great.  But as the official practice?  No way.

cmdremily, 2023-05-13:
The jumper has plastic covers everywhere and the conductive pins are inset in the housing. People drop them all the time and they're designed to not short stuff when dropped onto a motherboard etc.  As long as you don't actually push it onto the wrong pins you're fine even if it bumps something. And if you plug it into the wrong pins, you run the same risk when powering on the board anyway. But fair enough, please at least mention it as a possibility for people who are confident.

kubik369, 2023-07-02:
It might be good to leave it as it is simply because it teaches people good practices. Yes, it is ok to hot-plug it in this case, but then you get people hot-plugging stepper drivers, motors, or god-forbid, some heating elements, where it is either destructive, dangerous, or both.