VoronDesign/Voron-Documentation #406: Backing up printer configuration files to GitHub does not fully work

jaseiffert, 2024-01-21:
community/howto/EricZimmerman

Backing up printer configuration files to GitHub
https://docs.vorondesign.com/community/howto/EricZimmerman/BackupConfigToGithub.html

The problem is this part is not working:
Download the autocommit.sh file using the command below:
The wget command is not correct. URL is wrong as well as autocomit.sh is spelled wrong or Eric change the name at a later date.

I finally found Eric's instructions and finished getting my Voron Configuration setup and working.
EricZimmerman/Voron-Documentation/community/howto/EricZimmerman/BackupConfigToGithub.md

I would think you could update the page from Eric's instructions or just put a link to Eric's instructions so you always have the latest updates.

Thanks,

Jeffery

T4KUUY4, 2024-01-21:
Hi
the wget commands works just fine on my end. Did you copy it yourself or did you click the little "clipboard" icon in the command window? Do you have any error message popping up when executing it?

EricZimmerman, 2024-01-21:
not sure i am following.

at the url above, this command:

```
wget -O /home/pi/printer_data/config/autocommit.sh https://raw.githubusercontent.com/EricZimmerman/VoronTools/main/autocommit.sh
```

if we take the url, part

```
https://raw.githubusercontent.com/EricZimmerman/VoronTools/main/autocommit.sh
```

resolves and downloads just fine:

[image]

so if you could please verify you are actually running the exact command in the first code block above, that would be great. because i literally just copy pasted it from there as i write this.

searching for

`https://raw.githubusercontent.com/EricZimmerman/VoronTools/main/autocommit.sh`

also shows an exact match, as found in the wget command.

perhaps it is your first part of the command which is using a different path for the users directory?

i have no idea what you have username wise, so perhaps this would work for you?

`wget -O ~/printer_data/config/autocommit.sh ` 

for the first part. the full command would be

```
wget -O ~/printer_data/config/autocommit.sh https://raw.githubusercontent.com/EricZimmerman/VoronTools/main/autocommit.sh
```

thanks!

jaseiffert, 2024-01-21:
Last login: Sun Jan 21 12:10:02 2024 from 192.168.15.5
jefferys@voron:~ $  cd ~/printer_data/config
jefferys@voron:~/printer_data/config $  wget -O /home/pi/printer_data/config/autocommit.sh https://raw.githubusercontent.com/EricZimmerman/VoronTools/main/autocommit.sh
/home/pi/printer_data/config/autocommit.sh: No such file or directory
jefferys@voron:~/printer_data/config $

My bad on saying autocommit.sh was spelled wrong. The monitor has tiny letters and I'm old lol.

Using Putty on Windows 10.

jaseiffert, 2024-01-21:
This worked for me:
wget -O ~/printer_data/config/autocommit.sh https://raw.githubusercontent.com/EricZimmerman/VoronTools/main/autocommit.sh

I found it on here:
https://github.com/EricZimmerman/Voron-Documentation/blob/main/community/howto/EricZimmerman/BackupConfigToGithub.md

jaseiffert, 2024-01-21:
I couldn't login as pi. Tried all the passwords I could find to try so logged into the user I setup with Pi Imager.
I even tried that password with the pi user.

That probably why it wouldn't work now that you mentioned it.

jaseiffert, 2024-01-22:
I found out that there is no pi user on my machine. I'm guessing by using the new Pi imager that you can set your own admin user name so unless you use that to create the pi user you won't have one in the OS by default. They also say not to use the standard password if you do make a pi user.
https://www.raspberrypi.com/documentation/computers/getting-started.html#user

Since you are using: 

Change directories into your config directory, which is where all of your configuration files live
 cd ~/printer_data/config

I would change this:
 wget -O /home/pi/printer_data/config/autocommit.sh https://raw.githubusercontent.com/EricZimmerman/VoronTools/main/autocommit.sh

To this:
wget -O ~/printer_data/config/autocommit.sh https://raw.githubusercontent.com/EricZimmerman/VoronTools/main/autocommit.sh

This would match the "cd ~/printer_data/config used above"

This:  nano /home/pi/printer_data/config/autocommit.sh
and any other lines referencing the /home/pi/printer_data/config/ path should be changed to ~/printer_data/config/

As the pi user doesn't seem to be a given any more when using the Raspberry Pi Imager I believe that would work better for those that don't have a pi user or don't realize they don't have a pi user.

Anyway, you guys know a whole lot more about this than I do.

Thanks for your help.

Jeffery