VoronDesign/Voron-Documentation #410: potential bed crash with community/howto/clee/sensorless_xy_homing.md

tomlikesrocks, 2024-03-20:
it works fine if the printhead is already raised up from the bed and ready to go. 
But if you had a previous error, for example on QGL and had too many attempts, the printhead could be at Z=0 and if you home from there it can crash into the bed.
**Currently displaying:**
```
[homing_override]
axes: xyz
gcode:
  {% set home_all = 'X' not in params and 'Y' not in params and 'Z' not in params %}

  {% if home_all or 'X' in params %}
    _HOME_X
  {% endif %}
  
  {% if home_all or 'Y' in params %}
    _HOME_Y
  {% endif %}
  
  {% if home_all or 'Z' in params %}
    G28 Z
    G1 Z10
  {% endif %}
```

**I changed mine to:**
```
[homing_override]
set_position_z: 0
axes: xyz
gcode:    
    # Now lift Z safely
    G1 Z30 F600  # Lift Z by 30mm

    {% set home_all = 'X' not in params and 'Y' not in params and 'Z' not in params %}
 
    {% if home_all or 'X' in params %}
      _HOME_X
    {% endif %}
  
    {% if home_all or 'Y' in params %}
      _HOME_Y
    {% endif %}
  
    {% if home_all or 'Z' in params %}
      G90
      G1 X175 Y175 F12000  ## Change this.  Should home to the z endstop pin if you don't use TAP.  Also, this is for a Voron 2.4 350.  
      G28 Z   ## Home Z
      G1 Z10 F1200
    {% endif %}
```

keyboard-magpie, 2025-07-14:
The sensorless doc got a big update since you posted this, this can likely be closed.

tomlikesrocks, 2025-07-15:
Thanks for letting me know, seems like it has been fixed with:
  SET_KINEMATIC_POSITION Z=1 SET_HOMED=Z
  G1 Z4 F1200
in the homing.cfg

closed 2025-07-15 by tomlikesrocks