VoronDesign/Voron-Documentation #416: relative_reference_index in Bed Mesh with Relative Reference Index guide is obsolete

Tam-Lin, 2024-04-15:
The relative_reference_index option for [bed_mesh] is removed as of Klipper 20240215.  Trying to use the example provided will prevent Klipper from starting.

shiftingtech, 2024-04-15:
The only relative_reference_index example I can find is [this one]( https://docs.vorondesign.com/tuning/relative_reference_index.html) which clearly states at the top that it is only for older installs, and then links to the zero_reference_position version.

Tam-Lin, 2024-04-16:
True; however, there's a difference between keeping a depreciated option around and a completely removed one.  And it may be best to encourage people to use the latest Klipper; I'm not sure what the value of having an old install would be.

shiftingtech, 2024-04-16:
I really can't agree with you here

1. the documentation for the old option is already an obscure side-branch, that nobody should really end up in unless they are actively looking for it.  The main path already takes you to documentation that uses the current option

2. I'm not sure what the value of forcing people to upgrade is.  People should be upgrading because they want the new features, not because somebody decided to force their hand, and *delete documentation.

3.There are very legitimate reasons for people to decide to stay on whatever version of klipper they have working, given the complex challenges some people encounter surrounding firmware updates and the like.

closed 2024-04-25 by shiftingtech