VoronDesign/Voron-Documentation #417: Links not working on /community/howto/EricZimmerman  /SlicerAndPrintStart.md

jaseiffert, 2024-05-13:
These two links at the bottom of the page give 404 errors.

    slicer config: https://github.com/EricZimmerman/VoronConfigs/blob/master/Slicer.md
    PRINT_START macro: https://github.com/EricZimmerman/VoronConfigs/blob/master/macros/print_base/print_start.cfg

I couldn't find them but I didn't do an extensive search.

Thanks,

Jeffery

shiftingtech, 2024-05-13:
@EricZimmerman get in here!

EricZimmerman, 2024-05-13:
Easy fix. Updates path is

https://github.com/EricZimmerman/Voron24/blob/PreKlippain/macros/print_base/print_start.cfg

Do you want a PR or can you update it direct?

shiftingtech, 2024-05-13:
A pr would be great

EricZimmerman, 2024-05-13:
done

https://github.com/VoronDesign/Voron-Documentation/pull/418