VoronDesign/Voron-Documentation #423: Require tach fan on all new kits

Carlsans, 2024-08-08:
I don't know if you have any power there but it need to be said. I bought a kit from a company I wont disclose here. I went with a dragon hotend. Has you may know, kit fans are reputed to be the cheapest. After less than a week, my hotend fan failed. The hotend mount melted and the hotend forced by pression from the extruder crashed in the bed. The main issue there is that the mount will melt if the fan fail. For me that's a more than major issue, I left my printer for 30 minutes and here is the result : 
[image]
That's how bad the bed was scratched. The kirigami bed absorbed most of the shock even bending it.
With the way kit sellers are concerned with the fan problem (they are not). I think the best way to make all voron printers safe in that regard would be to propose kits that enforce the use of tach fans. Even the best fan can and will fail at some time.

nemgrea, 2024-08-08:
moved to documentation repo as the proposed fix does not involve a change to the v0 repo

EricZimmerman, 2025-03-15:
this would also require DIODES being supplied, since they are not on the boards by default